philseeley/boatinstrument #3: Issues with boatinstrument on CoreMP135

mgrouch, 2024-08-01:
Hello,

I've managed to build and run boatinstrument on CoreMP135 with flutter-pi.

The build was done on CircleCI using these scripts:

https://github.com/bareboat-necessities/boatinstrument/tree/main/.circleci

It was published to github 

https://github.com/bareboat-necessities/lysmarine_gen/releases/download/vTest/boatinstrument-0.0.1.1-flutterpi_arm32.tgz

Flutter-pi was installed on CoreMp135 using this script

````

sudo apt install cmake libgl1-mesa-dev libgles2-mesa-dev libegl1-mesa-dev \
 libdrm-dev libgbm-dev ttf-mscorefonts-installer fontconfig \
 libsystemd-dev libinput-dev libudev-dev  libxkbcommon-dev

sudo fc-cache

git clone --recursive https://github.com/ardera/flutter-pi

cd flutter-pi

mkdir build && cd build

cmake ..
make -j`nproc`

sudo make install

````

And running it:

````

wget https://github.com/bareboat-necessities/lysmarine_gen/releases/download/vTest/boatinstrument-0.0.1.1-flutterpi_arm32.tgz

gzip -cd boatinstrument-0.0.1.1-flutterpi_arm32.tgz | tar xvf -

flutter-pi --release ./boatinstrument

````

When I run it, it renders something but not everything

````

root@CoreMP135:~# flutter-pi --release ./boatinstrument
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
WARNING: Detected llvmpipe (ie. software rendering) as the OpenGL ES renderer.
         Check that flutter-pi has permission to use the 3D graphics hardware,
         or try running it as root.
         This warning will probably result in a "failed to set mode" error
         later on in the initialization.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Function not implemented
egl_gbm_render_surface.c: Will retry without modifiers

````

mgrouch, 2024-08-01:
[image]

mgrouch, 2024-08-02:
Also when I move mouse I see these errors:

````
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
````

philseeley, 2024-08-02:
Unfortunately I only have a headless RPi4 onboard and no HDMI monitor, so I'm unable to test the flutter-pi build at the moment. 

What kind of Graphics rendering does the CoreMP135 have? I wonder if one on the -dimensions, --pixelformat or --videomode options to flutter-pi might help.

Perhaps raising this with Hannes over at flutter-pi might be worth a shot.

I'll have access to an HDMI display etc in a couple of months.

Cheers
Phil

mgrouch, 2024-08-04:
I've tested arm64 linux version on Bareboat Necessites OS and it works on HDMI touchscreen monitor.

https://github.com/bareboat-necessities/lysmarine_gen/releases/download/vTest/boatinstrument-0.0.1.1-linux-aarch64.tgz

mgrouch, 2024-08-06:
@philseeley 

big issue on pi is that swipe to next screen using touch gestures works only rarely. 
most of the times it doesn’t. Switching between pages almost unusable right now. 
not sure where to troubleshoot that in code. 

Thanks

philseeley, 2024-08-06:
@mgrouch that's very disappointing. I'm using a standard Flutter GestureDetector widget at:

https://github.com/philseeley/boatinstrument/blob/80540e32307d6d3b876f67ac9052e57975b1f6b5/lib/main.dart#L120C13-L120C28

I can create a debug branch and put a bunch of print statements so we can track if it's detecting any inputs at all.

Would that work for you?

As I don't currently have a monitor on my Pi I'm unable to debug this locally, sorry.

mgrouch, 2024-08-06:
Yes I can try building and running version with debug statements. just please update
version in .yaml file for it. 

Thanks

philseeley, 2024-08-06:
@mgrouch looks like I can reproduce the fault using VNC to my headless Pi4. So I can have a look at what's going on.

Looks like the pointer and clicks are initiating callbacks, but the values passed aren't what are expected.

It'll take a few days to get time to have a look.

Thanks for finding this.

mgrouch, 2024-08-06:
@philseeley 

latest version of BBN OS imagine bundles 
your boatinstrument

you might play with it to see how it integrates with everything else. 

Download link

https://cloudsmith.io/~bbn-projects/repos/bbn-repo/packages/?q=lysmarine

thanks!

philseeley, 2024-08-10:
@mgrouch did you managed to get things working on the CoreMP135?

mgrouch, 2024-08-19:
There is a new suggestion on how to fix error with mouse on flutter-pi

https://github.com/ardera/flutter-pi/issues/437#issuecomment-2293663573

thanks

philseeley, 2024-11-02:
Hi @mgrouch,

I've built Debian packages that I hope will install and run on the CoreMP135.

I've only built then for Debian 12 bookworm.

Instructions on how to subscribe are here:

https://philseeley.github.io/debian-ppa/

If you just want the .deb files:

https://philseeley.github.io/debian-ppa/pool/main/f/flutter-pi/flutter-pi_1.0.0-d12_armhf.deb
https://philseeley.github.io/debian-ppa/pool/main/b/boatinstrument-fpi-generic/boatinstrument-fpi-generic_0.2.0-d12_armhf.deb

mgrouch, 2024-11-07:
Still same issue with touchscreen with versions 0.2.0 and with 0.3.0.3

````
root@CoreMP135:/usr/share/boatinstrument-fpi# ./flutter-pi --release .
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
WARNING: Detected llvmpipe (ie. software rendering) as the OpenGL ES renderer.
         Check that flutter-pi has permission to use the 3D graphics hardware,
         or try running it as root.
         This warning will probably result in a "failed to set mode" error
         later on in the initialization.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Function not implemented
egl_gbm_render_surface.c: Will retry without modifiers
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: SocketException: Connection failed (OS Error: No such file or directory, errno = 2), address = /run/user/0/bus, port = 0
#0      _NativeSocket.startConnect (dart:io-patch/socket_patch.dart:721)
#1      _NativeSocket.connect (dart:io-patch/socket_patch.dart:984)
#2      _RawSocket.connect (dart:io-patch/socket_patch.dart:1908)
#3      RawSocket.connect (dart:io-patch/socket_patch.dart:21)
#4      DBusClient._openSocket (package:dbus/src/dbus_client.dart:780)
#5      DBusClient._connect (package:dbus/src/dbus_client.dart:800)
#6      DBusClient.callMethod (package:dbus/src/dbus_client.dart:620)
#7      DBusRemoteObject.callMethod (package:dbus/src/dbus_remote_object.dart:188)
#8      WakelockPlusLinuxPlugin.toggle (package:wakelock_plus/src/wakelock_plus_linux_plugin.dart:41)
<asynchronous suspension>

window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
````

Ignore socket error.

This is the error of concern:

````
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
````

There is this issue still open:
https://github.com/ardera/flutter-pi/issues/437

But it has suggested workaround:

 Maybe this helps for now: 

https://gist.github.com/ardera/e7ddf7c8e477d258a26dfa9c40002626 (though you'll still get the error messages)

There are some might be relevant reports:

https://github.com/ardera/flutter-pi/issues/452

https://github.com/ardera/flutter-pi/issues/456

philseeley, 2024-11-07:
Having looked at the flutter-pi issues, it could be the display of the mouse cursor. I've applied the patch from https://github.com/ardera/flutter-pi/issues/452 and have built a flutter-pi package for you to test.

https://philseeley.github.io/debian-ppa/test/flutter-pi_1.0.0-d12_armhf.deb

mgrouch, 2024-11-07:
With this version I see different error and touchscreen still doesn't work:

````
root@CoreMP135:/usr/share/boatinstrument-fpi# ls -l /usr/bin/flutter-pi
-rwxr-xr-x 1 root root 345620 Nov  7 21:15 /usr/bin/flutter-pi
root@CoreMP135:/usr/share/boatinstrument-fpi# /usr/bin/flutter-pi --release .
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
WARNING: Detected llvmpipe (ie. software rendering) as the OpenGL ES renderer.
         Check that flutter-pi has permission to use the 3D graphics hardware,
         or try running it as root.
         This warning will probably result in a "failed to set mode" error
         later on in the initialization.
00:00:02.998  [libseat/backend/seatd.c:212] Enabling seat
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Function not implemented
egl_gbm_render_surface.c: Will retry without modifiers
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: SocketException: Connection failed (OS Error: No such file or directory, errno = 2), address = /run/user/0/bus, port = 0
#0      _NativeSocket.startConnect (dart:io-patch/socket_patch.dart:721)
#1      _NativeSocket.connect (dart:io-patch/socket_patch.dart:984)
#2      _RawSocket.connect (dart:io-patch/socket_patch.dart:1908)
#3      RawSocket.connect (dart:io-patch/socket_patch.dart:21)
#4      DBusClient._openSocket (package:dbus/src/dbus_client.dart:780)
#5      DBusClient._connect (package:dbus/src/dbus_client.dart:800)
#6      DBusClient.callMethod (package:dbus/src/dbus_client.dart:620)
#7      DBusRemoteObject.callMethod (package:dbus/src/dbus_remote_object.dart:188)
#8      WakelockPlusLinuxPlugin.toggle (package:wakelock_plus/src/wakelock_plus_linux_plugin.dart:41)
<asynchronous suspension>

libinput error: event2  - WingCool Inc. TouchScreen Mouse: client bug: event processing lagging behind by 629ms, your system is too slow
libinput error: event2  - WingCool Inc. TouchScreen Mouse: client bug: event processing lagging behind by 890ms, your system is too slow
````

mgrouch, 2024-11-08:
It looks like the issue is with touch gestures and some timeouts in their recognition logic.
Even with mouse enabled after mouse swipe and release the screen takes time to switch to next one.
Other error about "unsupported framebuffer stride" seems only has effect of mouse cursor not showing up.

I also see these errors:

````
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.ScreenSaver was not provided by any .service files
#0      DBusClient._callMethod (package:dbus/src/dbus_client.dart:1120)
<asynchronous suspension>
#1      DBusClient.callMethod (package:dbus/src/dbus_client.dart:621)
<asynchronous suspension>
#2      WakelockPlusLinuxPlugin.toggle.<anonymous closure> (package:wakelock_plus/src/wakelock_plus_linux_plugin.dart:47)
<asynchronous suspension>
#3      WakelockPlusLinuxPlugin.toggle (package:wakelock_plus/src/wakelock_plus_linux_plugin.dart:40)
<asynchronous suspension>

````

Thanks

mgrouch, 2024-11-08:
BMW and Toyota use Flutter-Embedded

https://www.bacancytechnology.com/blog/flutter-for-embedded

philseeley, 2024-11-12:
> I also see these errors:
>
> ````
> [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.ScreenSaver was not [...]

This is from the package used to keep the screen awake that requires some desktop integration, which isn't applicable when running under flutter-pi.

I've raised issue #19 for this.