philseeley/boatinstrument #5: Detect SignalK automatically using mDNS lookup

mgrouch, 2024-08-04:
Detect SignalK automatically using mDNS lookup

It's possible to do that so the user wouldn't need to go into settings to set SignalK URL.

Thanks

philseeley, 2024-08-05:
Thanks Mikhail, excellent idea.

philseeley, 2024-08-05:
First cut of the service discovery is committed. Took a bit of time to find out that the mDNS option hadn't been enabled
 in signalk server. I had assumed it was on as the hostname was being resolved by the client, but obviously this is done be a host-based service.

@mgrouch can you let me know if this works for BBN, thanks.

mgrouch, 2024-08-05:
@philseeley 

can you please increase the version from: 0.0.1+1
to higher on in pubspec.yaml
as GitHub doesn’t allow redeploy of same version ?

I will then merge, build on CircleCi and test. 

BBN OS has mDNS enabled on SignalK by default. 

Thanks

philseeley, 2024-08-05:
> can you please increase the version from: 0.0.1+1

Done.

mgrouch, 2024-08-05:
Yes it works on BBN. Shows data without need to configure SignalK connection (host).

If you need binaries, they are under 'Assets' here:

https://github.com/bareboat-necessities/lysmarine_gen/releases?q=test&expanded=true

Thanks

philseeley, 2024-08-06:
> Yes it works on BBN. Shows data without need to configure SignalK connection (host).

Excellent, thanks.

I've tested it successfully on Android, MacOS and Linux(Fedora), but there is an [open issue](https://github.com/flutter/flutter/issues/42102) for certain versions of iOS, where multicasting isn't allowed.

Windows is also not discovering the services, but then it already wasn't resolving local DNS names.

mgrouch, 2024-08-07:
They suggest to use Bonsoir library

> positive results from the use of https://pub.dev/packages/bonsoir instead of multicast_dns, while doing mDNS service discovery.
> Works both on iOS and Android.

philseeley, 2024-08-07:
The problem with the Bonsoir library and also some others is that they rely on platform specific code, e.g. the Apple Bonjour and Android NSD. So they have problems supporting other platforms.

The multicast_dns package implements the nsd service on the underlying Dart socket layer, so it's much more portable to all platforms. It is a permission issue that I had and solved on MacOS, but the iOS setup is missing the permissions file and it's not obvious how to add it. There's an open issue to document how to do it, which I'm watching.

philseeley, 2024-08-25:
So I tried to get iOS working with the multicast_dns package and that involved asking for permission from Apple to use multicast. And unsurprisingly that got refused. So I have another at the bonsoir plugin and it now supports all the OSs.

So I've converted over to that and tested it successfully on iOS, Android and MacOS.

closed 2024-08-25 by philseeley