philseeley/boatinstrument #8: Course data paths

mgrouch, 2024-08-06:
SignalK uses many different paths prefixes for course data.

When I activate "Navigate to Waypoint" in OpenCPN I see prefix

````
navigation.courseRhumbline
````

It seems come from APB, XTE, RMB sentences sent by OpenCPN
sources: OpenCPN.EC  (APB), OpenCPN.EC  (XTE), OpenCPN.EC  (RMB)

When I activate "Navigate to Waypoint" in FreeboardSK I see prefix

````
navigation.course
````

source: courseApi

When I activate course data in derived-data plugin I see prefix:

````
navigation.courseGreatCircle
````
and source: derived-data

When I activate Course Data Provider SK plugin the prefix is:

````
navigation.course.calcValues
````
and source 'course-provider'

I think it would be nice to make a prefix configurable for XTE, VMG, TTG, Next Waypoint Distance

Thanks!

mgrouch, 2024-08-07:
New signalK provides v2 deltas for course data

https://demo.signalk.org/documentation/whats_new.html

(at least it should)

with prefix

````
navigation.course
````

However code https://github.com/philseeley/boatinstrument/blob/main/lib/widgets/navigation_box.dart

is using 

````
navigation.courseGreatCircle
````

Thanks

philseeley, 2024-08-07:
@mgrouch will need some more of your help to test this as I only have a Raymarine autopilot setup.

What I've done as an initial idea is allowed wildcards in paths and replaces all the instances of 'navigation.courseGreatCircle.' with 'navigation.*.'.

Not sure this is all that's needed, but it would be good if all the autopilot sources produced the same ending values.

mgrouch, 2024-08-07:
They are not producing same values unfortunately. And it’s not clear for me 
how to differentiate the ones corresponding to
currently active autopilot route. 

Let me open issue for OpenCPN. 

I’ve already posted on forums

https://www.cruisersforum.com/forums/f134/opencpn-v5-9-4-beta-test-released-288251-3.html#post3923151

mgrouch, 2024-08-07:
I’ve opened this issue 

https://github.com/OpenCPN/OpenCPN/issues/4085

mgrouch, 2024-08-08:
What I think is also needed is a setting 
for course data with expression to match SignalK
‘source’ as well. 
could be on a main settings page. 
Then your changes to match SignalK path and
additional (optional) match of SignalK source will make it flexible enough. 
SignalK source is also in SignalK json feed (just different field). 

Thanks!

philseeley, 2024-08-09:
I've looked into the configuration of a "source" for Boxes before and feel that the use of "Source Priorities" in signalk is a better way to go. That way the preferred source is defined once and all displays referencing the data will display consistent values.

For the varying/inconsistent paths issue, these are the ones I'm seeing related to navigation and the autopilot:
```
navigation.courseGreatCircle.crossTrackError
navigation.courseGreatCircle.nextPoint.distance
navigation.courseGreatCircle.nextPoint.timeToGo
navigation.courseGreatCircle.nextPoint.velocityMadeGood
navigation.courseOverGroundTrue
navigation.currentRoute.waypoints
navigation.headingTrue
navigation.magneticVariation
navigation.position
navigation.speedOverGround
steering.autopilot.state
steering.autopilot.target.headingMagnetic
steering.autopilot.target.headingTrue
steering.autopilot.target.windAngleApparent
```
Can you detail where you see different paths for equivalent values. I can then think of a good way to deal this this, either through a set of specific Boxes or a configuration setting. Hopefully there will be some future signalk/plugin changes to rationalise the set of paths used.

mgrouch, 2024-08-09:
````
steering.autopilot.state
steering.autopilot.target.headingMagnetic
steering.autopilot.target.headingTrue
steering.autopilot.target.windAngleApparent
```

Those are not actual headings or angles. Those are commands to autopilot to steer.

philseeley, 2024-08-10:
I've been trying to get the 'Course Data provide' plugin to create some paths, but I've not managed to get anything. Would this be because I'm just using the Raymarine MFD to set the course?

Note, I'm not using the 'Derived Data' plugin for the 'Course Data', so the paths I'm seeing like 'navigation.courseGreatCircle.crossTrackError' must be being generated from the N2K feed from my YDEN gateway.

Basically I'm very confused about how all his is supposed to integrate :o(

mgrouch, 2024-08-10:
There are other ways too. 
Activate ‘navigate to waypoint’ in OpenCPN. 
Activate ‘navigate to waypoint’ in FreeboardSK

and they will make SignalK to generate paths which are not consistent. (Not only different sources but different paths). 

I think SignalK has an issue with way it’s implemented. 

There is SignalK discord channel. It’s better to ask there 

https://discord.gg/uuZrwz4dCS

philseeley, 2024-08-11:
I've checked that I'm running the latest version 2.8.3 and the v2 apis aren't present. So it looks like the v2 apis are a work in progress. Reading the v2 docs, it looks like they realise the historic piecemeal implementation of things like routes and autopilot wasn't sustainable and are addressing it.

So it's a wait and see for the time being.

If there's specific paths that would make the Boxes work better for your setup let me know.

philseeley, 2024-08-11:
Actually I can see some v2 apis, but the course values are not getting set for me. I'll see if I'm missing something.

mgrouch, 2024-08-11:
>If there's specific paths that would make the >Boxes work better for your setup let me know.

A box for tides and currents harmonics would be very helpful. 
It could check if harmonics file exists on a file system and based on the location data from gps find data for next 24 hours and display it.

philseeley, 2024-08-13:
More digging into tides and it seems there are many formats and many sources, some of which have gone away due to copyright issue.

There's already a long list for [OpenCPN](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:supplementary_software:tides), so I think it's not something I'll be adding anytime soon, sorry.