philseeley/boatinstrument #11: Boatinstrument doesn't close audio on anchor alarm after it is stopped.

mgrouch, 2024-08-22:
Boatinstrument doesn't close audio on anchor alarm after it is stopped.

There is some kind of leak.

Thanks

philseeley, 2024-08-25:
I've recently updated all the dependancies and the audioplayers plugin went up a major version. So this might have addressed this.

mgrouch, 2024-09-05:
There is still issue with sound (I think). 
Not sure if it’s boatinstrument. However there are lot of audio streams from boatinstrument that keep showing as not closed. 
Looks like there some kind of leak. 

Thanks

philseeley, 2026-08-15:
I've been testing the latest version with pulseaudio and can see connections coming and going, but can't see anything lingering.

So hopefully the audio package I'm using has fixed this.

closed 2026-08-15 by philseeley