philseeley/boatinstrument #13: Handle notifications in the background.

philseeley, 2024-08-25:
Currently the encapsulation of each Box means that functionality like the Anchor Alarm will only be activated if it's displayed on the current page.

Also we need to allow the app to run in the background.

mgrouch, 2024-08-25:
I've tested 0.11.1. 
I'm using my own config file for it with SignalK host set to 127.0.0.1 and mdns Disabled.
Somehow of first start version 0.11.1 replaced 127.0.0.1 with something like mypi.local and ignored my preset.

Thanks

mgrouch, 2024-08-25:
I see what happened.

New format is
````
  "signalkHost": "127.0.0.1",
  "signalkPort": 3000,
````

Old one

````
  "signalkServer": "127.0.0.1:3000",
````

I'll change it on my side.

Thanks

philseeley, 2024-09-11:
Implemented by commit f1d008677667bce0a70695eab6e2b2b5cb72dbc0.

closed 2024-09-11 by philseeley