philseeley/boatinstrument #14: Engine/Tanks levels/Power generation widgets

mgrouch, 2024-10-28:
It would be useful to add 
Engine/Tanks levels/Power generation widgets

Here are some ideas for layouts from another project 

[image]
[image]
[image]

Thanks!

philseeley, 2024-10-28:
Hi @mgrouch,

Can you confirm the signalk paths you're seeing and their units?

Do they conform to the signalk spec?

Picture 1:
- RPM - propulsion.\*.revolutions
- Degrees C - propulsion.\*.temperature
- psi - propulsion.\*.oilPressure

Picture 2:
- Battery % - electrical.batteries.\*.capacity.stateOfCharge or electrical.batteries.\*.capacity.remaining
- PV - electrical.solar.\*.voltage and electrical.solar.\*.current
- AC Load - electrical.inverters.\*.dc.voltage and electrical.inverters.\*.dc.current
- DC Load - electrical.batteries.\*.voltage and electrical.batteries.\*.current

Picture 3:
- Grey - tanks.wasteWater.\*.currentLevel
- Black - tanks.blackWater.\*.currentLevel
- Fresh - tanks.freshWater.\*.currentLevel
- Fuel - tanks.fuel.\*.currentLevel
- Lube - tanks.lubrication.\*.currentLevel

mgrouch, 2024-10-28:
They might not be compliant to the spec. 
If people take them from Victron they are on different paths. So I think you can code with SignalK defaults but let user to override paths. 

Thanks

mgrouch, 2024-11-07:
I've looked at 3.0.3.0

Few more would be nice:

- Exhaust Temperature (in Engine group)
- Fridge Temperature (in Environment)
- Battery Bank Temperature (in Electrical)
- Solar Current (in Electrical)
- Shore Power Current (in Electrical)
- Rate of Turn (in Boat)
- Fuel Burn Rate (in Engine)
- Bilge Level

Thanks for the great work on this!

philseeley, 2024-12-14:
For the digital switching, do you have any kit like an EmpirBus to test with?

philseeley, 2025-02-13:
@mgrouch think I've got most of this done.

closed 2025-02-13 by philseeley