philseeley/boatinstrument #18: Provide default config file with pre-configured widgets

mgrouch, 2024-11-07:
Provide default config file with pre-configured widgets.

I've made this one while working on Bareboat Necessities OS:

boatinstrument.json

````
{
  "version": 0,
  "valueSmoothing": 1,
  "discoverServer": false,
  "signalkHost": "127.0.0.1",
  "signalkPort": 3000,
  "signalkMinPeriod": 500,
  "signalkConnectionTimeout": 20000,
  "dataTimeout": 10000,
  "demoMode": false,
  "darkMode": true,
  "wrapPages": true,
  "brightnessControl": false,
  "keepAwake": true,
  "autoConfirmActions": false,
  "pageTimerOnStart": false,
  "distanceUnits": "nm",
  "m2nmThreshold": 500,
  "speedUnits": "kts",
  "windSpeedUnits": "kts",
  "depthUnits": "ft",
  "temperatureUnits": "c",
  "airPressureUnits": "millibar",
  "oilPressureUnits": "kpa",
  "portStarboardColors": "redGreen",
  "pages": [
    {
      "name": "one",
      "pageRows": [
        {
          "percentage": 0.9167567661830357,
          "columns": [
            {
              "percentage": 0.3735430029374135,
              "rows": [
                {
                  "percentage": 0.66128249040301,
                  "boxes": [
                    {
                      "percentage": 1.0,
                      "id": "wind-rose",
                      "settings": {
                        "type": "normal",
                        "showLabels": true,
                        "showButton": false,
                        "autoSwitchingDelay": 15
                      }
                    }
                  ]
                },
                {
                  "percentage": 0.17593341286171132,
                  "boxes": [
                    {
                      "percentage": 0.5,
                      "id": "wind-apparent-angle",
                      "settings": {}
                    },
                    {
                      "percentage": 0.5,
                      "id": "wind-speed-apparent",
                      "settings": {}
                    }
[... 673 lines cut ...]
          ]
        }
      ]
    }
  ],
  "boxSettings": {
    "date-time": {
      "dateFormat": "yyyy-MM-dd"
    },
    "autopilot-control": {
      "clientID": "boatinstrument-autopilot-1"
    },
    "anchor-alarm": {
      "clientID": "boatinstrument-anchor-alarm-1",
      "recordSeconds": 10,
      "recordPoints": 1000
    }
  }
}
````

I takes a while for a user to set up widgets. I think it's better to start from reasonable default.

Thanks
--MG

philseeley, 2024-11-11:
If we did this, I think we'd need different default configs for phones and tablets or portrait vs landscape.

I think I would favour the option of starting as is, but give an option to load an example config if users want that

mgrouch, 2024-12-04:
At least for Linux it might be better to have some starting configuration.
It's a effort to create. It's easier to remove unneeded things from something existing and add few customizations.

I maintain this one for Bareboat Necessities (BBN) OS

https://github.com/bareboat-necessities/lysmarine_gen/blob/bookworm/install-scripts/6-desktop/files/boatinstrument.json

Thanks

philseeley, 2025-01-13:
I've added some very basic one page configs for landscape and portrait modes.

**Note:** if you're building an SD Card, then dropping a config into “/boot/boatinstrument.json” will update the config on the next boot. See [docs](https://philseeley.github.io/debian-ppa/docs/boatinstrument/main.html).

closed 2025-01-13 by philseeley