philseeley/boatinstrument #30: Wind instrument displays 120 degrees as 12

mgrouch, 2025-01-15:
[image]

Thanks

mgrouch, 2025-01-15:
Or as 20 on the right side

philseeley, 2025-01-15:
Hi Mikhail, well spotted! But this is actually as designed, as I'm rendering the speed boxes on top of the rose so as to maximise the size of the speed boxes.

I prioritise the apparent wind needle and position the speed boxes appropriately, but don't move them unless really necessary. I then prioritise the speed boxes over the true wind needle, so you will find some edge cases where the true wind is significantly different from the apparent where the speed boxes render over the true wind needle. 

This ensures that whilst sailing the apparent wind direction and speed are always as clear as possible.

You can also remove the speed boxes if you'd rather show speeds separately.

closed 2025-01-15 by philseeley