philseeley/boatinstrument #32: login in embedded webapp does not work (on android phone)

Colug0, 2025-01-24:
if I try to login in an embedded webapp (grafana or signalk webapp) the keyboard shows for a split second and disappears right away. So no login is possible.

philseeley, 2025-01-24:
Hi @Colug0,

Thanks for raising this. The support for embedded web pages is a bit experimental and I'm using the [flutter_inappwebview](https://pub.dev/packages/flutter_inappwebview) package for this, but unfortunately this has some outstanding issues with respect to keyboard interactions: 

https://github.com/pichillilorenzo/flutter_inappwebview/issues/1918
https://github.com/pichillilorenzo/flutter_inappwebview/issues/137

flutter_inappwebview is supposed to support all platforms, but unfortunately I've found it's lacking in some/most.

I'll have a look at switching to the [webview_flutter](https://pub.dev/packages/webview_flutter) which only supports iOS and Android, but I think that's OK as on other platforms using a real browser would always be better.

FYI one of the reasons for this app was because other solutions were based around browsers or other embedded HTML frameworks and I never found they gave a good user experience.

philseeley, 2025-01-24:
Hi @Colug0,

Well that wasn't very successful. The webview_flutter package exhibits the same symptoms as the flutter_inappwebview.

Looking through the various issues, I think the problems is that when the keyboard is displayed the app page is resized and therefore the widget tree is rebuilt. This causes the WebView to reload the webpage. There may be ways around this for an app that's just skinning a WebView, but for this Boat Instrument there's lots more going on with multiple pages and Boxes.

There is also an issue where the page can't be scrolled. Again this might not be a problem for a browser skinning app, but is for me.

So... unfortunately I can't see a way of making this work like a real browser would.

I am adding experimental support for a VNC viewer Box which "might" be more successful, but YMMV.

Colug0, 2025-01-24:
Thanks for the reply. Maybe it is the easiest to disable authentication ;-)

Do you have a documentation which Box uses which path in signalk?

philseeley, 2025-01-25:
A path to box map is a great idea, thanks. 

Web authentication can be a pain. We have used Grafana a lot to track things like wind speed trends and have found it very useful to add annotations of things like sail changes. Unfortunately adding annotations required write permission, so you can change the dashboards as well. Others have also suggested that annotations should have a separate permission or be included in the read permission, but not sure that will ever get done.

One of the reasons for the Web Box was to try and show Grafana graphs, but I've not managed to get it to work well, so I'm now adding support for Graph Boxes.

So far I've added graphs for true and apparent wind, water and air temperature, atmospheric pressure, speed and SOG.

Any other graphs you'd find useful?

mgrouch, 2025-01-25:
Power usage, solar input would be useful 

Thanks

philseeley, 2025-02-17:
> Power usage, solar input would be useful
> 
> Thanks

Done.

philseeley, 2025-02-17:
> Do you have a documentation which Box uses which path in signalk?

Hi @Colug0, Box to Path Mappings are now documented [here](https://philseeley.github.io/debian-ppa/docs/boatinstrument/box-path-mappings.html)

As for the Web Authentication, that's not going to get fixed unfortunately.

closed 2025-02-17 by philseeley