philseeley/boatinstrument #36: Max Min and Smoothing Algorithms

mgrouch, 2025-02-19:
For calculating Max/Min over sliding window there is efficient algorithm:

https://github.com/EvanBalster/STL_mono_wedge  based on this paper

https://arxiv.org/abs/cs/0610046

I use implementation in C++ which you can probably easily translate into Dart asking DeepSeek AI:

https://github.com/bareboat-necessities/bbn-wave-period-esp32/blob/main/bbn_wave_freq_m5atomS3/MinMaxLemire.h

And for smoothing you can use Kalman filter smoothing:

https://github.com/bareboat-necessities/bbn-wave-period-esp32/blob/main/bbn_wave_freq_m5atomS3/KalmanSmoother.h

Thanks

mgrouch, 2025-02-19:
Here are AI translations:

MonoWedge:

Below is the Dart implementation of the provided C++ code for the monotonic wedge algorithm. The implementation includes the `mono_wedge_search`, `mono_wedge_update`, and convenience functions for min and max wedges.

```dart
// Dart implementation of the monotonic wedge algorithm

import 'dart:collection';

/// Search routine used to determine deletion range in `mono_wedge_update`.
///
/// Similar to `std::lower_bound`, returns the first element in the range for which
/// `comp(value, element)` returns false.
///
/// The range must be sorted with regard to `comp`.
/// Complexity is below log2(N) with respect to wedge size.
/// Facilitates amortized constant complexity in `mono_wedge_update`.
Iterator<T> monoWedgeSearch<T>(
    Iterator<T> begin, Iterator<T> end, T value, bool Function(T, T) comp) {
  var size = end - begin;
  if (size <= 0) return end;

  // Linear search through at most J elements, where J = log2(N-J).
  var searchPos = end - 1;
  var i = 1;
  for (; ((size - i) >> i) > 0; ++i, --searchPos) {
    if (comp(searchPos.current, value)) return searchPos + 1;
  }
  // Afterwards run a binary search
  return _lowerBound(begin, searchPos + 1, value, comp);
}

/// Binary search implementation similar to `std::lower_bound`.
Iterator<T> _lowerBound<T>(
    Iterator<T> begin, Iterator<T> end, T value, bool Function(T, T) comp) {
  var low = begin;
  var high = end;
  while (low != high) {
    var mid = low + ((high - low) >> 1);
    if (comp(mid.current, value)) {
      low = mid + 1;
    } else {
      high = mid;
    }
  }
  return low;
}

/// Update a monotonic wedge with a new value.
///
/// Erases values which do not satisfy `comp(element, value)`, then
/// appends `value` to the wedge via `push_back`.
///
/// Complexity is less than log2(N) with respect to wedge size.
/// Complexity of N calls is O(N), if the wedge is initially empty.
/// Thus, amortized complexity over many calls is constant.
void monoWedgeUpdate<T>(
    List<T> wedge, T value, bool Function(T, T) comp) {
  var i = monoWedgeSearch(wedge.iterator, wedge.iterator + wedge.length, value, comp);
  wedge.removeRange(i.index, wedge.length);
  wedge.add(value);
}

/// Convenience function for updating a min-wedge.
void minWedgeUpdate<T>(List<T> wedge, T value) {
  monoWedgeUpdate(wedge, value, (a, b) => a < b);
}

/// Convenience function for updating a max-wedge.
void maxWedgeUpdate<T>(List<T> wedge, T value) {
  monoWedgeUpdate(wedge, value, (a, b) => a > b);
}

/// Example usage
void main() {
  // Example: Min-wedge
  var minWedge = <int>[];
  minWedgeUpdate(minWedge, 3);
  minWedgeUpdate(minWedge, 1);
  minWedgeUpdate(minWedge, 2);
  print(minWedge); // Output: [1, 2]

  // Example: Max-wedge
  var maxWedge = <int>[];
  maxWedgeUpdate(maxWedge, 3);
  maxWedgeUpdate(maxWedge, 1);
  maxWedgeUpdate(maxWedge, 2);
  print(maxWedge); // Output: [3, 2]
}
```

### Key Points:
1. **Dart Iterators**: Dart's `Iterator` is used to simulate the behavior of C++ iterators. The `monoWedgeSearch` function works with iterators to maintain compatibility with the original algorithm.
2. **Binary Search**: The `_lowerBound` function implements a binary search similar to `std::lower_bound`.
3. **List Operations**: Dart's `List` is used as the wedge container, and `removeRange` and `add` are used to modify the wedge.
4. **Comparator Functions**: The comparator functions (`comp`) are passed as arguments to determine the order of elements in the wedge.
5. **Convenience Functions**: `minWedgeUpdate` and `maxWedgeUpdate` are provided for ease of use with min and max wedges.

This implementation should behave similarly to the original C++ code. Let me know if you need further clarification or adjustments!

mgrouch, 2025-02-19:
Certainly! Below is the Dart implementation of the provided C++ code for the `STL_RingBuffer` using the previously implemented `MonoWedge` functionality. The `STL_RingBuffer` is reimplemented as a Dart class, and the `MonoWedge` functions are integrated into it.

### Dart Implementation

```dart
import 'dart:collection';

/// RingBuffer implementation in Dart
class RingBuffer<T> {
  final List<T?> _buffer;
  int _head = 0;
  int _tail = 0;
  int _size = 0;
  final int _capacity;

  RingBuffer(int capacity)
      : _capacity = capacity,
        _buffer = List<T?>.filled(capacity, null);

  /// Returns the number of elements in the buffer
  int get size => _size;

  /// Returns the maximum capacity of the buffer
  int get capacity => _capacity;

  /// Checks if the buffer is empty
  bool get isEmpty => _size == 0;

  /// Checks if the buffer is full
  bool get isFull => _size == _capacity;

  /// Adds an element to the end of the buffer
  void pushBack(T value) {
    if (isFull) {
      throw Exception("RingBuffer is full");
    }
    _buffer[_tail] = value;
    _tail = (_tail + 1) % _capacity;
    _size++;
  }

  /// Removes and returns the element at the front of the buffer
  T popFront() {
    if (isEmpty) {
      throw Exception("RingBuffer is empty");
    }
    T value = _buffer[_head]!;
    _buffer[_head] = null;
    _head = (_head + 1) % _capacity;
    _size--;
    return value;
  }

  /// Returns the element at the front of the buffer without removing it
  T front() {
    if (isEmpty) {
      throw Exception("RingBuffer is empty");
    }
    return _buffer[_head]!;
  }

  /// Returns the element at the back of the buffer without removing it
  T back() {
    if (isEmpty) {
      throw Exception("RingBuffer is empty");
    }
    return _buffer[(_tail - 1 + _capacity) % _capacity]!;
  }

  /// Clears the buffer
  void clear() {
    _head = 0;
    _tail = 0;
    _size = 0;
    _buffer.fillRange(0, _capacity, null);
  }

  /// Returns an iterator for the buffer
  Iterator<T> get iterator => _RingBufferIterator(this);

  /// Helper class for iterating over the RingBuffer
  class _RingBufferIterator implements Iterator<T> {
    final RingBuffer<T> _ringBuffer;
    int _index;
    int _count = 0;

    _RingBufferIterator(this._ringBuffer) : _index = _ringBuffer._head;

    @override
    T get current => _ringBuffer._buffer[_index]!;

    @override
    bool moveNext() {
      if (_count >= _ringBuffer._size) {
        return false;
      }
      _index = (_index + 1) % _ringBuffer._capacity;
      _count++;
      return true;
    }
  }
}

/// Monotonic Wedge implementation in Dart
class MonoWedge<T> {
  final RingBuffer<T> _wedge;
  final bool Function(T, T) _comp;

  MonoWedge(int capacity, this._comp) : _wedge = RingBuffer<T>(capacity);

  /// Updates the wedge with a new value
  void update(T value) {
    var i = _monoWedgeSearch(_wedge.iterator, _wedge.size, value, _comp);
    _eraseFrom(i);
    _wedge.pushBack(value);
  }

  /// Searches for the position to insert the new value
  Iterator<T> _monoWedgeSearch(
      Iterator<T> begin, int size, T value, bool Function(T, T) comp) {
    if (size <= 0) return begin;

    // Linear search through at most J elements, where J = log2(N-J)
    var searchPos = begin;
    var i = 1;
    for (; ((size - i) >> i) > 0; ++i) {
      if (comp(searchPos.current, value)) return searchPos;
      searchPos.moveNext();
    }
    // Binary search for the rest
    return _lowerBound(begin, searchPos, value, comp);
  }

  /// Binary search implementation
  Iterator<T> _lowerBound(
      Iterator<T> begin, Iterator<T> end, T value, bool Function(T, T) comp) {
    var low = begin;
    var high = end;
    while (low != high) {
      var mid = low + ((high - low) >> 1);
      if (comp(mid.current, value)) {
        low = mid + 1;
      } else {
        high = mid;
      }
    }
    return low;
  }

  /// Erases elements from the given position to the end
  void _eraseFrom(Iterator<T> position) {
    while (position.moveNext()) {
      _wedge.popFront();
    }
  }

  /// Returns the front element of the wedge
  T front() => _wedge.front();

  /// Returns the back element of the wedge
  T back() => _wedge.back();

  /// Clears the wedge
  void clear() => _wedge.clear();
}

/// Convenience function for updating a min-wedge
void minWedgeUpdate<T>(MonoWedge<T> wedge, T value) {
  wedge.update(value);
}

/// Convenience function for updating a max-wedge
void maxWedgeUpdate<T>(MonoWedge<T> wedge, T value) {
  wedge.update(value);
}

/// Example usage
void main() {
  // Example: Min-wedge
  var minWedge = MonoWedge<int>(10, (a, b) => a < b);
  minWedgeUpdate(minWedge, 3);
  minWedgeUpdate(minWedge, 1);
  minWedgeUpdate(minWedge, 2);
  print(minWedge.front()); // Output: 1

  // Example: Max-wedge
  var maxWedge = MonoWedge<int>(10, (a, b) => a > b);
  maxWedgeUpdate(maxWedge, 3);
  maxWedgeUpdate(maxWedge, 1);
  maxWedgeUpdate(maxWedge, 2);
  print(maxWedge.front()); // Output: 3
}
```

### Key Points:
1. **RingBuffer**:
   - A circular buffer is implemented using a fixed-size list (`_buffer`) and indices (`_head` and `_tail`) to manage the buffer's state.
   - Supports `pushBack`, `popFront`, `front`, `back`, and `clear` operations.
   - Includes an iterator for traversing the buffer.

2. **MonoWedge**:
   - Uses the `RingBuffer` as the underlying container.
   - Implements the `mono_wedge_search` and `mono_wedge_update` logic from the C++ code.
   - Provides `update`, `front`, `back`, and `clear` methods.

3. **Comparator Functions**:
   - The comparator (`_comp`) is passed to the `MonoWedge` constructor to determine the order of elements (min or max wedge).

4. **Convenience Functions**:
   - `minWedgeUpdate` and `maxWedgeUpdate` are provided for ease of use with min and max wedges.

This implementation should behave similarly to the original C++ code. Let me know if you need further clarification or adjustments!

mgrouch, 2025-02-19:
```
// This code implements a one-dimensional Kalman filter and is meant to be used in a Dart environment.

class KalmanSmootherVars {
  // Kalman filter variables
  double processNoise; // process noise covariance
  double measurementUncertainty; // measurement uncertainty
  double estimationUncertainty; // estimation uncertainty
  double kalmanGain; // kalman gain
  double filteredValue; // value

  KalmanSmootherVars(this.processNoise, this.measurementUncertainty, this.estimationUncertainty);
}

void kalmanSmootherInit(KalmanSmootherVars kalmanVars, double processNoise, double sensorNoise, double estimatedError) {
  kalmanVars.processNoise = processNoise;
  kalmanVars.measurementUncertainty = sensorNoise;
  kalmanVars.estimationUncertainty = estimatedError;
}

void kalmanSmootherSetInitial(KalmanSmootherVars kalmanVars, double initialValue) {
  kalmanVars.filteredValue = initialValue;
}

double kalmanSmootherUpdate(KalmanSmootherVars kalmanVars, double measurement) {
  // measurement update
  kalmanVars.kalmanGain = kalmanVars.estimationUncertainty / (kalmanVars.estimationUncertainty + kalmanVars.measurementUncertainty);
  double currentEstimate = kalmanVars.filteredValue + kalmanVars.kalmanGain * (measurement - kalmanVars.filteredValue);
  kalmanVars.estimationUncertainty = (1.0 - kalmanVars.kalmanGain) * kalmanVars.estimationUncertainty + (kalmanVars.filteredValue - currentEstimate).abs() * kalmanVars.processNoise;
  kalmanVars.filteredValue = currentEstimate;
  return kalmanVars.filteredValue;
}

```

philseeley, 2025-02-20:
Thanks @mgrouch, excellent information!

I wasn't aware of Kalman filtering so I came up with a smoothing method of my own, but I can now see there's already a [Dart package](https://pub.dev/packages/simple_kalman) for this.

I'll give the Kalman smoothing a go, but not sure if we'd gain much from doing min/max over a sliding range that the graphs don't already give.

philseeley, 2025-02-20:
I updated the Kalman package to take an initial value and then ran and graphed some test wind speed data in m/s in the range 4.94 to 9.57. The Kalman filter was initialised with an errorMeasure=10 and errorEstimate=5:
```
import 'package:simple_kalman/simple_kalman.dart';

import 'package:boatinstrument/boatinstrument_controller.dart';

void main() {
  List<double> data =[
6.95, 7.05, 7.05, 7.36, 6.54, 7.1, 5.82, 6.18, 6.59, 6.85, 6.07, 6.59, 6.54, 6.74, 6.13, 6.64, 6.18, 6.28,
\<snip\>
5.77, 6.13, 5.97, 6.64, 5.56,
  ];

  double? smooth, kalman;
  SimpleKalman simpleKalman = SimpleKalman(errorMeasure: 10, errorEstimate: 5, q: 0.1, initValue: data[0]);

  print(',smooth,kalman,raw');
  for(int i=0; i < data.length; ++i) {
    double d = data[i];
    smooth = averageDouble(smooth??d, d, smooth: 20);
    kalman = simpleKalman.filtered(d);
    print('$i,$smooth,$kalman,$d');
  }
}
```
I ran tests with multiple q values against the existing smoothing.

[image]
[image]
[image]
[image]
[image]

philseeley, 2025-02-21:
I observed the following:

- "20 smth 1.0 cal" - Using my maximum smoothing and minimum Kalman smoothing the results are almost identical.
- Using anything more than minimal Kalman, i.e. < 1.0, the results are a little too smooth.
- The results vary depending upon the errorMeasure used, see below, and therefore this would need to be tuned to the specific data being used, i.e. AWS vs SOG.
- Tuning the errorMeasure could be problematic if the range could change, e.g. it might have to be different for light and strong winds.

My gut feeling is that Kalman smoothing is more suited to statistical analysis when multiple runs with varying values can be compared before the results can be relied upon. Or situations where it's applied to specific data with well known and analysed bounds.

What I've currently implemented seems to cover a broad range of data and appears "good enough" for the intended application. Perhaps the requirement is more for "damping" than for "data smoothing".

Will definitely be keeping Kalman smoothing in my back pocket for future use, thanks.

[image]
[image]

closed 2025-02-21 by philseeley