philseeley/boatinstrument #37: Error building in circleci

mgrouch, 2025-03-18:
````
SLVR:         fact: no versions of actions_menu_appbar from path match 0.0.1
SLVR:         derived: not actions_menu_appbar 0.0.1 from path
SLVR:         fact: actions_menu_appbar from path doesn't exist (could not find package actions_menu_appbar at "../actions_menu_appbar")
SLVR:         conflict: actions_menu_appbar from path doesn't exist (could not find package actions_menu_appbar at "../actions_menu_appbar")
SLVR:         ! actions_menu_appbar from path is satisfied by actions_menu_appbar from path
SLVR:         ! which is caused by "boatinstrument depends on actions_menu_appbar from path"
SLVR:         ! thus: version solving failed
````

Thanks

philseeley, 2025-03-18:
Hi Mikhail,

The build is now dependant on a widget I use in other apps.

I see you're pulling my updates into your own clone for a CI build. I do have a CI build in a separate repo for performing releases.

I think any CI in this repo should be performing tests rather than release builds. Flutter has a GUI test framework for this, but I  haven't had the time to work out how to use it yet.

Can I ask why you need to do your own builds rather than using the repos I'm publishing? Are you doing some testing?

I'd like to make things as simple as possible for you to include the app in BBN releases. 

Cheers
Phil

mgrouch, 2025-03-18:
It’s mostly for historical reasons. I’ve been loading boatinstrument on BBN OS before you came up with Debian packages. Also I think you do not build arm32 flutter deb package which I would need for CoreMP135 version of BBN OS.

philseeley, 2025-03-18:
I do build arm32 packages for the flutter-pi versions:

https://philseeley.github.io/debian-ppa/pool/main/f/flutter-pi/flutter-pi_1.0.1-d12_armhf.deb
https://philseeley.github.io/debian-ppa/pool/main/b/boatinstrument-fpi-generic/boatinstrument-fpi-generic_0.6.0-d12_armhf.deb

If these don't work for you but you have working builds then what am I missing?.

FYI: the package releasing repo is https://github.com/philseeley/package-release

mgrouch, 2025-03-18:
Yes I should start using your .deb packages

https://github.com/bareboat-necessities/lysmarine_gen/issues/502

closed 2025-03-24 by philseeley