philseeley/boatinstrument #38: Allow signalk server url to be specified without port

JamesPerrins, 2025-03-24:
If the url could be specified without a port, the app could be used remotely to monitor the boat systems using cloudflares zero tunnel which requires the port to be specified within the cloudflare setup.

At the moment it fails because we have to specify the port

philseeley, 2025-03-24:
Hi James,

It might work if you specify 80 for the port.
 
If not then would you also need HTTPS support?

What platform are you running on? 

We'll definitely get this to work👍

Cheers
Phil

philseeley, 2025-03-24:
Done... Deleted the comment👍

philseeley, 2025-03-25:
Hi James,

I've just tested with your hostname and the port set to 80 and got connected ok.

If I use https I get:

```
HandshakeException: Handshake error in client (OS Error: 
	WRONG_VERSION_NUMBER(tls_record.cc:231))
#0   BoatInstrumentController._discoverServices (package:boatinstrument/boatinstrument_controller.dart:678:9)
[E] Error discovering services
```

JamesPerrins, 2025-03-25:
Hi Phil,

I changed the line in boatinstrument_controller.dart to

Uri uri = Uri(scheme: _settings!.demoMode ? 'https' : (port == 443 ?
'https' : 'http') , host: host, port: port, path: '/signalk');

And that seems to work with https so long as the port is set to 443

Looking at securing it - cloudflare has lots of options - but thinking
setting up a service application token is probably the easiest (most of the
rest are more interactive) -
https://developers.cloudflare.com/cloudflare-one/identity/service-tokens/

Best wishes
James

JamesPerrins, 2025-03-25:
Second service setup:

boatsecure.xxxxxx (same domain as other)
Cloudflare token or headers
{"CF-Access-Client-Id":"2d0443ce2a62adc8d42647103811c088.access",
"CF-Access-Client-Secret":"17765c1ea756aac92dc6782ac9de923fac1283003f63e195f413a4f43a16583c"}

philseeley, 2025-03-25:
Ah, yes of course! So yes http and https work.

What I now need to do is decide the best way to support a more flexible configuration. For example there may be setups that have additional or a different path section of the URL.

philseeley, 2025-03-25:
I can easily add the Cloudflare tokens as headers in the service discovery part, but WebSocket.connect doesn't allow headers to be specified.
```
      _channel = WebSocketChannel.connect(wsUri.replace(query: 'subscribe=none'));
```
So looks like this would require lots more looking at, which won't be a quick job at the moment.

I think it might need a replacement of the Class used to handle the websocket, which would need lots of testing to ensure that didn't break other functionality.

JamesPerrins, 2025-03-25:
Agreed - as you say looks like the generic web socket doesn't support it.

I think IOWebSocketChannel does which appears to support android, ios and
windows but not web.
I've done a quick test and its a simple swap for Windows at least

*http.Response response = await http.get(uri,headers:
{'CF-Access-Client-Id':'abc.access','CF-Access-Client-Secret':'xyz'}
).timeout(const Duration(seconds: 10));*

and then

*_channel = IOWebSocketChannel.connect(wsUri.replace(query:
'subscribe=none'), headers:
{'CF-Access-Client-Id':'abc.access','CF-Access-Client-Secret':'xyz'});*

I'll see if I can get an android build going too

Failing that maybe https://pub.dev/packages/websocket_universal

But as you say - all requires refactoring and testing for any knock ons.

philseeley, 2025-03-26:
That's a great start, thanks. 

There will be a few other places where http.get is called, e.g. for static data and the switches and autopilot boxes. 

Looks like we'd need to add an httpGet to the controller for all these to use, then the way of setting the headers needed.

philseeley, 2025-03-26:
I've  done a first cut using IOWebSocketChannel and refactored all the http connections into the controller.

Currently I'm forcing https and adding your headers for boatsecure.xxxxxx. Providing a way to specify these setting would be the next step.

I've checked this in on the "http-headers" branch if you'd like to check this out.

JamesPerrins, 2025-03-27:
Thanks - I'll have a look.

I'd hacked something similar (but more crudely) as a test - and it all
seemed to be working OK (not able to test autopilot or switches)
Still learning though - discovered authentication has to be admin for
anchor which confused me for a while

Also had a slight battle building android version - due largely to my
inexperience with flutter I suspect.
I had to drop the flutter version to 3.27.4 to avoid build errors, and I
had a few challenges working out how signing worked - but got there in the
end and its working on and android phone OK

Tested windows and android

philseeley, 2025-03-27:
Great work.

I've updated the "http-headers" branch to deal with a SignalK URL rather than separate host and port. When reading the settings it should migrate to the new format.

Could you give it a try, thanks.

Note that it will probably default to http in the migration as you couldn't previously use https.

philseeley, 2025-03-27:
I think the admin requirement for the autopilot might be because it saves the anchor location in its plugin settings file, so that they persist over a reboot.

But I've not confirmed that with Scott yet.

JamesPerrins, 2025-03-27:
Yep - that seems to be working thanks
As you said - migration didn't include the https, but did include port 443
appended

So I changed http to https and dropped the :443 and it all seems to work
(from a quick test on widows build)

philseeley, 2025-03-28:
Getting close... You can now add HTTP headers from the settings page.

JamesPerrins, 2025-03-28:
That's brilliant Phil,

Just pulled and merged - all seems to work on windows and android - as
before can't test autopilot or switches - but otherwise looking great.

Many thanks

philseeley, 2025-04-29:
Hi @JamesPerrins, I've merged the http-headers branch, so this will be in the next release.

closed 2025-04-29 by philseeley