philseeley/boatinstrument #41: Tanks display

mgrouch, 2025-05-01:
The tanks % didn't show up automatically. I had to go into settings and select tank ID for each chart.
Average boating user wouldn't be able to figure that out as they expect it to work plug and play.

Thanks

mgrouch, 2025-05-01:
Also color coded tank content (bar) would look better.

I propose

Fresh Water - blue
Black Water - red
Fuel - green
Lube - yellow
Grey Water - orange

Thanks

philseeley, 2025-05-05:
Hi @mgrouch,

Colours all done, nice idea thanks. I checked they're all good in the Dark/Light/Night modes.

The auto setting of the tank ID is a little tricky due the the class hierarchy in place, so for now I've added a message to set the ID. The widget classes are setup for efficiency by having fixed/final data, so on construction the ID must be known. After choosing in the setting page the widget is recreated with the fixed/final ID.

Hope that makes some sense.

closed 2025-05-05 by philseeley